erica-synths/diy-eurorack #2: Unzip for better version control

wraybowling, 2020-07-16:
If there happen to be any downstream improvements, the diffs will be impossible to see. I recommend these projects remain uncompressed in simple folders. Thanks

truedat101, 2021-03-31:
It seems like somebody outside of Erica should make a fork of the repo (I looked and there are indeed 44 forks), but thus far, there has only been a single contribution.  I think I have a fork.  I will look into making some of these easier repository changes like unpacking all of the components in my fork.  I thought it was unusual to just upload zips, as if the open source dump was done in a hurry.

truedat101, 2021-03-31:
I looked through about 20 of the forks and there were no improvements, no one has touched their forks, probably partially because the way the open source dump got done, it makes contribution somewhat difficult.