erica-synths/diy-eurorack #5: MIDI-CV source code

holmesrichards, 2020-10-20:
I'm very pleased you have open sourced this hardware. It would be good to do likewise with the MIDI-CV software. While you do provide the .hex file, that's of little use if one wants to address bugs or add new functionality. Please consider making the .ino/c++ source code available.

truedat101, 2021-03-31:
Yes, part of open source is providing the software.  I did a search to see if there are any alternative firmwares out there but I did not find anything for this module.

forestcaver, 2021-03-31:
It really isnt an obligation. They’ve kindly open sourced the hardware - this allows for repairs or others to learn from the design.

The software may (for all I know) contain proprietary non-open source code. It may contain intellectual property they want to protect. They may not want the support headache of people opening issue on github or asking for support for the code.

Choosing not to open source the code is their decision and I am happy with whatever they choose. They are under no obligation to release it!

Please, let’s just be grateful for what they have released rather than hassling them for what they have chosen not to release (please :-) )

truedat101, 2021-03-31:
Software is presumably a big part of the crown jewels and trade secrets of a modular synth maker.  As someone who has worked in open source for years and worked with companies who are taking proprietary art (patents, software, methods, hardware) open, companies doing so are only obligated to follow the law in the jurisdictions where they operate.  That said, companies are releasing open source for a reason.  I am assuming the reason for the dump was to provide some level of support for the end of life DIY modules.  Great.  Nobody is zinging the company Erica for that.  This is a useful start.

Going forward, to provide a second life for these modules, people want to hack on what has been provided (bug fixes, enhancements).  The promise of, and expectation of the open source community is that they can enhance, modify, inspect, use, distribute, sell, whatever is released in open source within the confines of the license applied to what has been released.  I think that is fair.

If Erica doesn't want to hear feedback, they can shut off Issues.  But I believe they've left this open as a sort of "support" for the EOL DIY modules that may still be out there.  Having an open source community takes work.  It provided headaches.  I can only speculate about the goals of the company.   I am happy to see they have provided this repository.