erica-synths/diy-eurorack #7: test point voltages on DIY VCO

laerm0, 2021-01-19:
Hi. I recently finished building the VCO, just in time for support to end. 🤦 I'm getting no sound out of it so I'd like to check the voltages to figure out where it went wrong. Can someone please post the voltages at the test points? Thanks.

hippertronics, 2021-03-31:
I've just built a V2.2 pcb version - I'll make an assumption that the test points are labelled the same. It works perfectly as far as I can tell. All pots and switches centred except FM. Trimmers on the main board centred, calibration carried out. No jacks connected. Measuring DC with a fluke multimeter.

TP1 +3V
TP2 -1.25v
TP3 -11.32V
TP4 -10.63V
TP5 +7.03V
TP6 +4.85v
TP7 -0.43V
TP8 +10.17V
TP9 -10.24V

laerm0, 2021-04-15:
Wow, my values are way off...

TP1 +3.07V
TP2 9.2v
TP3 0.67V
TP4 11.34V
TP5 6.81V
TP6 1.42v
TP7 5.57V
TP8 +10.15V
TP9 0.775V

TP1 and 8 and close enough but the rest, yikes. I didn't get a single negative value, which seems like a clue. Any guesses where that could be going wrong?