erica-synths/diy-eurorack #11: BASSLINE main pcb error

KoSv, 2021-01-29:
I'm not able to uplaod the main gerber file to my pcb maker. I tryied merging the drill files, but no success. thanks

nelsonmilum, 2021-02-23:
Decompress DIYBASS2_Gerber, there are two folders "Control" and "Main" zip them individually and upload them separately to jlcpcb like that

KoSv, 2021-02-23:
> Decompress DIYBASS2_Gerber, there are two folders "Control" and "Main" zip them individually and upload them separately to jlcpcb like that

Thanks for the response! I did exactly this, of course! 
But when I upload the main PCB it doesn't show up as a 3D model on their website (like other pcb's normally do ). Also The PCB size doesn't get auto recognized and so on. How do I get around this? Do you have the experience uploading to jlcpcb files like that? greets ! : )

nelsonmilum, 2021-03-07:
So I think that because the PCB in question has a notch cut out of one corner for the power header, the Gerber loading software that JLCPCB uses gets confused. I just used the dimensions of the Control PCB and ignored the lack of preview. Since after you submit a job they do have a process to validate that there's no issues. I should have a set of these arriving in the next couple of days. If it's successful or not I'll let you know :-)

infinitemachinery, 2021-03-07:
I encountered this same issue at JLCPCB with the Bassline main PCB. Their gerber viewer chokes on the files but the PCB will be manufactured to specs. Enter the board dimensions, 79mm x 106mm, upload the gerber files and specify any other manufacturing options. The board will come out just fine.

KoSv, 2021-03-08:
good to know! many thanks : )

closed 2021-03-08 by KoSv