erica-synths/diy-eurorack #15: Incorrect BOM and build guide for BASSLINE

TheConceptBoy, 2023-06-25:
Good day.

I believe that the build guide and BOM for the Erica Synth BASSLINE module are not up to date. The Board files appear to be a version 2 and contain resistors and dip pins that are neither present in the build guide nor the BOM downloaded from the official Erica Github.

[image]

Fore example resistors R177 and R178 are not marked as NU and have values next to them but neither the BOM nor the build guide have any mention or reference to these resistors.  Same with R179 and R180 and a bunch more.
[image]
[image]

Is there a build guide and updated bom available for the version 2 somewhere?

MLTQ, 2023-07-26:
Similar problem here- zip layouts are conflicting. 
The BOM indicates that R93, R118, and R131 are 33k1%, while the values board layout indicates that they are 56k1%, 39k1%, and 22k1%. Given that luckily there is only one resistor in the BOM for each of those values, (there is no 22k at all), I'm going to assume the BOM is correct here and thus the values layout should be updated.

TheConceptBoy, 2023-07-26:
> Similar problem here- zip layouts are conflicting. The BOM indicates that R93, R118, and R131 are 33k1%, while the values board layout indicates that they are 56k1%, 39k1%, and 22k1%. Given that luckily there is only one resistor in the BOM for each of those values, (there is no 22k at all), I'm going to assume the BOM is correct here and thus the values layout should be updated. EDIT: R94 is listed in the BOM as a 100k1%, but on the values board it is 220k1%. There is no 220k1% in the BOM, so I'm assuming this is a typo as well.

It is worth noting that at the top of the BOM, it says Version 1 but at the top of the board view with values it says Version 2. I had actually populated the missing transistor and resistors from the version 2 board view file and most of the module started working. 

My only dread now is that I must figure out what differences the board view and the BOM have.. might have to go through each and every component and figure out by color code if it's correct.

So in the nutshell. I think the BOM is outdated. Maybe it makes sense in my case because my physical PCB says it's a V2 so I should probably be following the board view pdf that also says v2

MLTQ, 2023-07-26:
Oh no... I've just nearly finished populating based on the BOM, luckily I've marked all the differences and have spare parts but that's a pretty critical problem.
Thanks for the update.

TheConceptBoy, 2023-07-26:
> Oh no... I've just nearly finished populating based on the BOM because thats what came with my kit, luckily I've marked all the differences and have spare parts but that's a pretty critical problem. Thanks for the update.

Once you get through it, please let us know if you got it working and everything runs ok. I gotta do some investigative debugging and value validation on mine.

MLTQ, 2023-07-26:
Tested it with the BOM resistors... definitely something wrong, extremely high pitched, filter didn't work at all. Replaced them with the values resistors and the sound seems more appropriately pitched, but the filter doesn't work still.

I'm guessing that that is an unrelated issue. 

If you could give me some values of test points it would be a huge help- TP37 & 36 show no response on the scope to twisting the filter cutoff dial. Not sure what I should be seeing at TP28 and 32, but those are the closest to where I think the issue would be!

TheConceptBoy, 2023-07-26:
The filter didn't work in my case because I had a missing transistor
somewhere closer to the bottom right of the main board.

davcon12, 2023-07-27:
On the schematics it has some voltages at the test points and after doing some maths I am coming close to those voltages but they might be voltages from a working module. Using 5% resistors these could be different voltages, up to 5% + or -. On 3 TPs I'm more than that difference. I am going by the board layout values.

MLTQ, 2023-07-27:
Those test voltages only confirm that the oscillator is working- doesn't tell me anything about the filter unfortunately.

davcon12, 2023-07-27:
I'm trying to figure this thing out myself. I had started on it and found discrepancies with the BOM, values board and schematics, so I guess that the best thing to do is to do all of the maths and work it out. 
I understand that your filter isn't working so have you checked to see if you have power to all your chips and in the right place. Also if you have bad or incorrect polarities, when that counts, components?